goodrobots/devFrame #1: Integrate retractable landing gear

fnoop, 2018-01-13:
eg. Tarot TL65B44

 - Looks pretty(er)
 - Makes people go 'ooh'
 - Removes legs out of way of cameras, sensors etc
 - Makes handling better (less swinging weight)

cglusky, 2018-01-13:
I have a set on order so I can test.

Likely best method is to use a plate between them and use antivibe mounts to clamp to bottom box tubes.  

You can see how they are doing it for these...

https://www.banggood.com/HML650-Retractable-Folding-Landing-Gear-For-Tarot-650-680pro-HMF-S550-p-941639.html

cglusky, 2018-04-29:
I have these in the shop.  I will design and test a mount but not sure it makes sense to use these in the D500 build.  Might make most sense to just add it to the wiki as an option.  They are heavy and I don't think most people will use them.

cglusky, 2020-01-07:
For d500 which is on hold until people say they want it.  Many 450 class frames out there to use.

closed 2020-01-07 by cglusky