goodrobots/devFrame #19: Rebuild D300 with OmniBusF4v2

cglusky, 2018-04-29:
ArduCopter 3.6 is in beta and support for small race class FC's is included.  I have an OmnibusF4v2 Pro in the shop along with BLHeli ESC's.  This gear is likely a better match to digital FPV use case on D300.

I will finish the current build with PixFalcon; test fly; Autotune; and document.  Then pull it apart and repeat with OmniBus et al with ArduCopter 3.6.

I also do not have step by step pics of the current build so this will provide an opportunity to fix docs.

This will also put the docs in a alignment with latest ArduCopter, which is a big change (more 4.0 than 3.6 seems to me), so we do not need to repeat right after we complete docs for 3.5.

cglusky, 2018-05-12:
D300 initial build and docs will be done with OmniBusF4 FC.  Running cleanflight with a RsPi and Raspi Cam module.  FPV will be done with ezwifibroadcast and DroneBridge intitially.

Once there is more time then AC3.6 and Maverick wifibc can be documented as well.

fnoop, 2018-05-12:
Boo!

cglusky, 2018-05-12:
Understand.  I don't like it much either.  Maverick is not limiting factor.  It's my lack of time to try to figure out AC3.6 and OminBus.

The good news is the D500 build gets bumped up.  I just want to put the D300 to bed and get the D500 going so I can have a bit more room to strap on gear and play with Maverick, FrsKy telemetry, and WifiBC.

Can always circle back around to D300 once OmniBus support is figured out in AC.

cglusky, 2018-05-20:
My D300 build will be with betaflight and FPV gear.  Will leave 300 class builds with ArduPilot etc to @fnoop  and others.

closed 2018-05-20 by cglusky