goodrobots/devFrame #36: Update all mounts to new 5mm spaced pilot holes

cglusky, 2019-03-14:
RC1 will use 1mm pilot holes vs through holes on end clamps and cross mounts.  All mounts need to be changed to support new spacing.

cglusky, 2020-01-08:
All mounts updated to new spacing

fnoop, 2020-01-08:
Commit?

What's the benefit/change of the new pilot holes?

cglusky, 2020-01-08:
Just standardizing everything going forward.  Had new designs with 15mm (essentially half of the 30mm made popular by small autopilots, which is actually 30.5 but I don't worry about the .5mm since printing and drilling have bigger tolerances) so they won out.  Any old mounts you have will still work as it's just a 1mm delta.  Maybe need a light ream with a drill bit but they will still work.

commit 1587b4d1 (2020-01-09): Antenna Mount Flat

15mm spacing and closes #36
closed 2020-01-12 by cglusky