goodrobots/devFrame #39: Test Brass Screws For GPS Mount

cglusky, 2020-01-08:
Test brass screws with mRO GPS/compass combo to see if we have fewer compass issues

fnoop, 2020-01-08:
What about nylon screws?

cglusky, 2020-01-08:
Good point.  Do you have any issues sourcing?  I have some here in shop but not sure they will be right sizes.  mRobo GPS cover is designed to be screwed in and capture GPS on the mount at same time.  But you could drill it out and use nylon bolt and nut.  The mount itself is designed to be screwed into roll cage but again you could use small nylon bolt/nut combo.

fnoop, 2020-01-08:
Yup easy peasy to get on ebay, most standard M sizes.  Also much easier to cut to size.

cglusky, 2020-01-09:
M2 nylon on order for testing.

cglusky, 2020-01-14:
If you use nylon they need to be M3 for required strength.

closed 2020-01-14 by cglusky