goodrobots/devFrame #97: See if copper tape is doing anything

cglusky, 2020-02-25:
using copper tape on bottom of GPS mount.  could also be al foil. aspirin or vitamin? 

https://ardupilot.org/copter/docs/common-compass-setup-advanced.html#compassmot-compensation-for-interference-from-the-power-wires-escs-and-motors

fnoop, 2020-02-26:
I used copper tape to shield the bottom of gps mount with minimal success.  Turns out much more important to shield radiators (usb3 plugs are a massive radiator).  Note if you shield the gps mount it should be connected to the gps groundplane to be really effective.
You can get EMI shielding material which is much more effective but it's quite expensive.

cglusky, 2020-02-26:
if i have time i will see if it makes any measurable difference to ground/drain the shielding.

cglusky, 2020-04-24:
issues like this likely best served via a blog post and any relevant results then baked back into docs.