goodrobots/devFrame #98: Consider embedding 3D part views into docs

SamuelDudley, 2020-04-16:
Github has a nice 3D viewer for stl files as per this screen shot
[image]
https://github.com/goodrobots/devFrame/blob/master/stl/Frame/devFrame%20Landing%20Gear%20Spacer.stl

it looks like this viewer can be embedded in a webpage:
https://help.github.com/en/github/managing-files-in-a-repository/3d-file-viewer#embedding-your-model-elsewhere

Perhaps this would be useful to add into the docs?

cglusky, 2020-04-16:
That was original plan.  Concerned it would become a performance issue if we used it more than once per page. And if I break out a separate page per part I start to recreate what already exists on repo.

SamuelDudley, 2020-04-16:
Okay, no worries!

closed 2020-04-16 by SamuelDudley