goodrobots/devFrame #106: jekyll bundle version

fnoop, 2020-04-23:
When trying to build the odcs with jekyll, getting this error:
```
Notice: /Stage[main]/Maverick_web::Maverick_docs/Exec[build-devframe_docs]/returns: /usr/lib/ruby/2.5.0/rubygems.rb:289:in `find_spec_for_exe': can't find gem bundler (>= 0.a) with executable bundle (Gem::GemNotFoundException)
Notice: /Stage[main]/Maverick_web::Maverick_docs/Exec[build-devframe_docs]/returns: 	from /usr/lib/ruby/2.5.0/rubygems.rb:308:in `activate_bin_path'
Notice: /Stage[main]/Maverick_web::Maverick_docs/Exec[build-devframe_docs]/returns: 	from /usr/local/bin/bundle:23:in `<main>'
Error: '/usr/local/bin/bundle exec jekyll build' returned 1 instead of one of [0]
Error: /Stage[main]/Maverick_web::Maverick_docs/Exec[build-devframe_docs]/returns: change from 'notrun' to ['0'] failed: '/usr/local/bin/bundle exec jekyll build' returned 1 instead of one of [0]
```

The issue is explained here:
https://bundler.io/blog/2019/05/14/solutions-for-cant-find-gem-bundler-with-executable-bundle.html

Doing a system update of gem (`gem update --system`) is not a good solution - it seems to update some fundamental components enough that it sends puppet absolutely haywire and completely breaks `maverick configure`.  Updating ruby is not a sensible option on most systems.   The easiest fix is simply to remove the BUNDLED version in Gemfile.lock:
```
BUNDLED WITH
-   2.1.2
```

cglusky, 2020-04-23:
Can do that for each versioned release.  Just going to come back with each commit for working branch.  And I am working from master for docs.  Does that work for you?

fnoop, 2020-04-23:
These comments suggest to bump the version up to something high, but that doesn't work.
https://github.com/rubygems/bundler-features/issues/88#issuecomment-109220888
https://github.com/rubygems/bundler/issues/3697#issuecomment-108100899

I guess the only option is either to keep manually removing it from the lockfile, or fix a version of bundler that doesn't get changed for as long as possible, so we can document it for other contributors, and can be set as a parameter in maverick manifest for the devframe docs.

Seems like a really strange thing to do - reading the comments above and lots of other similar ones not one that a lot of ruby developers agree with..

cglusky, 2020-04-23:
seems to be restrictive.  and not clear if there is a reason, which never makes things better.  right now my last local build totally locked up.  and all i did was add a few images.  so have to get that fixed first.

cglusky, 2020-04-23:
https://stackoverflow.com/questions/18383402/how-to-specify-minimum-bundler-version-for-gemfile

don't know if that helps

fnoop, 2020-04-24:
It's not a minimum version that's needed - it's the *exact* version.  I had bundler 2.1.4 installed (the latest, I think, currently), but it needed exactly 2.1.2 to work.  Any other version produced the same cryptic error.