goodrobots/devFrame #107: Some files not added to .gitignore

fnoop, 2020-04-23:
After a build, git status shows:
```
[dev] [mav@maverick-ubuntuvm ~/software/devframe_docs/docs]$ git status
On branch master
Your branch is up-to-date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   _site/index.html

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	../.jekyll-cache/Jekyll/Cache/Jekyll--Cache/b7/9606fb3afea5bd1609ed40b622142f1c98125abcfe89a76a661b0e8e343910
	../.jekyll-cache/Jekyll/Cache/Jekyll--Converters--Markdown/b3/d457b1d68e5ff968c4542ca290b32924c454353eb8e356d21ce0b29c76dc09
	../.jekyll-cache/Jekyll/Cache/Jekyll--Converters--Markdown/ec/6705e243521f16569cd2e6ff86f9c3b212ca2a1c90d8a99de4a1e8da8053f8
```

cglusky, 2020-04-23:
every time you build _site should regen.  what are we missing?

fnoop, 2020-04-23:
_site is the local finished product, not?  It shouldn't be committed back to git?

cglusky, 2020-04-23:
yes, true since ghpages compiles..  let me see about adding _site dir

cglusky, 2020-04-23:
it's already in my gitignore so not sure what's going on.

cglusky, 2020-04-23:
this should be fixed.  i added gitignore tweaks after i had already committed a bunch of stuff.

closed 2020-04-23 by cglusky