goodrobots/devFrame #108: jekyll install/build steps

fnoop, 2020-04-23:
Aside from #106, there are further gems missing when it comes to trying to build with jekyll, in a typical OS install (eg. ubuntu).
It seems the easiest way to resolve this is to use bundler to install the gems locally in docs/vendor/bundle, somewhat akin to virtualenv in python.
```
sudo gem install bundler --version 2.1.2
git clone https://github.com/goodrobots/devframe
cd devframe/docs
bundle config set deployment 'true'
bundle install --deployment
bundle exec jekyll build
```

cglusky, 2020-04-23:
used this method.  https://jekyllrb.com/tutorials/using-jekyll-with-bundler/

so perhaps we need a better gemfile as vendor folder is ignored on purpose.

why would people be working on devframe docs using a companion computer?  or are you just trying to serve a PWA?

fnoop, 2020-04-24:
1) For an offline/centralised copy of the docs - I'm starting to collate a repo of relevant docs starting with ardupilot, px4, maverick and now devframe, so they're available in the field or when offline.
2) As a pre-configured area for contribution that needs less expertise/setup by a potential contributor.  All these compiled docs - devframe with jekyll, ardupilot with sphinx, are quite difficult to setup and a real barrier for any potential contributor to get started.

cglusky, 2020-04-24:
well, we should probably config a PWA some time in future.  there are jekyll pwa plugins.  but before i go there i would like to look at reducing the overhead for creating docs.