goodrobots/devFrame #109: Absolute URLs hardcoded into markup

fnoop, 2020-04-24:
When trying to load the compiled site through nginx, it fails because links/images/references are all hardcoded with '/devFrame' prefix.  URLs should ideally be relative, so the site can be hosted anywhere.
[image]

cglusky, 2020-04-24:
not hardcoded.  pulls from config.yaml.  via tokens inserted into liquid template syntax.  

`[build section]({{ site.baseurl }}/docs/build-intro/) `

`# Site 
locale: "en-US"
title: "devFrame docs"
description: >- # this means to ignore newlines 
  "devFrame is an Open Source quadcopter and rover frame design." 
url: "https://goodrobots.github.io"
baseurl: "/devFrame"
stlurl: "/blob/master/stl/"`

fnoop, 2020-04-24:
Ah!  Cool ok, I'll look into that, either making it relative or perhaps adding a custom config.yaml for maverick.  Is there a particular reason you've moved it to /devFrame rather than . or /?

cglusky, 2020-04-24:
did not want to hog root as it probably should be some sort of index to goodrobots projects.  eventually.  my understanding is to host from root of github org you have to create a separate repo as well.

cglusky, 2020-04-24:
likely be best to run a custom config yaml in case someone does want to create a PR from maverick devframe docs.  the baseurl property really helps for local dev so you can easily mirror githib setups.